technoblogy/ulisp-esp #11: Please provide an (UNWIND-PROTECT x) function

phmarek, 2019-02-23:
I'd like to have my ESP32 available via WIFI all the time.
I've got a simple telnet REPL working; the one problem left is that some error (any error) breaks the Lisp REPL and returns to the serial console.

I'd imagine that `UNWIND-PROTECT` would set some variable to the current execution point (and croak if it's already set, to make it non-reentrable); `error` and `error2` would check that variable, and when it's set store the current error in another variable (that can be queried from Lisp) and instead of doing the `longjmp(exception, 1)` they should restore the execution point to the saved one.

So my REPL could just run something like

```lisp
(setf *last-error* nil)
(let ((result (unwind-protect (eval input))))
  (cond
    (*last-error*
      (princ "Error: " client)
      (princ *last-error* client))
    (t
      (princ result))))
```

technoblogy, 2019-04-11:
Sorry for the delay in replying. I agree that **unwind-protect** would be useful, but the action "restore the execution point to the saved one" is difficult because of the need to restore the state of the stack. I'll give it some more thought, but I suspect it will involve a major rewrite of uLisp.

technoblogy, 2019-06-24:
Still thinking about this!

dragoncoder047, 2022-08-15:
Whether it ends up in official uLisp or not, I found that @Goheeca and @m-g-r implemented `(unwind-protect)` and a few other error-related functions. See these two links:

* https://github.com/Goheeca/redbear_duo-uLisp/commit/4894c13 
* http://forum.ulisp.com/t/error-handling-in-ulisp/691/7

You'll have to apply the patches yourself. I did on my uLisp, and it works great.

technoblogy, 2023-03-29:
Now included in uLisp Release 4.4.

closed 2023-03-29 by technoblogy