technoblogy/ulisp-esp #18: RAM usage - pls clarify

stacksmith, 2019-06-12:
I am curious about WORKSPACESIZE being set to 8000... I am just figuring my way around ESP32, but it appears that there is 520k of SRAM and lots of flash...
Could you explain why it is not 64K or something like that?
Thanks

technoblogy, 2019-06-12:
Bear in mind that it's 8000 Lisp cells, each of which is 8 bytes, so it's actually 32Kbytes. By all means experiment with larger sizes, but I think that was what I could get to work reliably; to use more of the SRAM you may have to create custom linker definitions, or allocate memory differently eg with malloc..

stacksmith, 2019-06-12:
I tried 16K and id did not compile.  I didn't have time to investigate; I thought I'd ask if there is some hard limit placed by you somewhere, like a table for GC or something...

stacksmith, 2019-06-12:
Oh, do you know if Arduino or any underlying infrastructure requires all code to be in RAM?

technoblogy, 2019-06-12:
As far as I know there are no fixed limits imposed by uLisp, although it is optimised for small-RAM systems.

Kaef has forked the ESP version of uLisp and extended it to work with the 4 Mbytes of RAM available on ESP-WROVER-32 modules; see:

https://github.com/Kaef/ulisp-esp

stacksmith, 2019-06-12:
Thanks a lot!

closed 2019-06-12 by stacksmith