technoblogy/ulisp-esp #19: Anonymous function recursion broken in 2.7b

brown131, 2019-06-13:
I am using the ESP8266 implementation of uLisp 2.7b. When I upgraded from 2.6a one of my functions broke with an "undefined function" error. It worked in 2.6.

Here is a simple function that recreates the issue. As you can see the name of the anonymous function ("fn") is no longer in scope in the function body.
```
(defun incx ()
  (let* ((x 0)
	 (fn (lambda ()
	       (incf x)
	       (if (> x 5)
		   x
		   (fn)))))
    (fn)))
```
```
>  (incx)
Error: 'fn' undefined
```

technoblogy, 2019-06-14:
Thanks - I'm investigating a similar problem at the moment and hope to have a solution soon. David

technoblogy, 2019-06-14:
Actually, I've just realised that in your simplified example the first call to (fn), in (fn) itself, is the one causing the error. The same happens on Common Lisp on my Mac.

The problem I'm investigating is that, under some circumstances in 2.7, a GC causes a variable to become undefined, and I imagine this is the problem you found when upgrading. Please uncomment:

    #define printgcs

and see if the error occurs just after a GC.

brown131, 2019-06-14:
Yes, sorry I meant the body of the fn function not the incx function body. I should have made that clearer.

I turned on printgcs and changed my code a little so it can run, and I think I was able to re-create the error I think you're talking about.

Here is a gist with the code that I ran:
https://gist.github.com/brown131/da274ce61639a9356cdf614c954df67f

Here is the output:
```
{5}2320>   (parse-json test)
{2085}
{2091}
{2111}
{2069}
{2052}
{2049}
{2032}
{2067}
{2047}
{2052}
{2040}
{2030}
{2050}
{2036}
{2051}
{2034}
{2044}
{2041}
{2044}
Error: 'json' undefined
```
Unless there's a bug in my code,  "json" should not be undefined.

technoblogy, 2019-06-14:
It looks like the same problem I'm trying to solve. Does your code work fine in uLisp 2.6?

brown131, 2019-06-15:
Yes it does. It works when I run it using 2.6a.

technoblogy, 2019-06-15:
OK, if it's a problem stick to 2.6a until I find a solution.

closed 2019-06-20 by technoblogy

technoblogy, 2019-06-21:
Hi, I think I've fixed this issue with Version 2.7c. I'll be interested to know if it fixes the problem you encountered with your code. Thanks, David

brown131, 2019-06-21:
Thanks for fixing that. I'll try it out this weekend and let you know if I still have a problem.

commit 0889d646 (2019-06-20): Version 2.7c - 20th June 2019

Fixes #19