technoblogy/ulisp-esp #29: REPL on the serial interface evaluates s-expressions as soon as syntactically complete

elbow, 2020-02-22:
Hi,

I presume this is "as-designed", but I was surprised that when I connect in to an esp32 on the usb-serial port, then the "repl" seems to evaluate s-expressions immediately as soon as they are syntactically complete.

IE if I type (+ 1 2) then as soon as I enter the closing ) the result is printed.

If I type 4 and then hit space the same thing happens.

Is this normal or did I do something strange in my setup?

Thanks

technoblogy, 2020-02-22:
That's normal, and is the way Lisp usually works. If you want to provide a line editor, to allow deleting/editing what you've entered, you would need to do this before passing the line to uLisp, as I did with my Lisp Badge:

http://www.ulisp.com/show?2L0C

technoblogy, 2020-04-29:
This is now fixed by the new lineditor and vt100 compile options in uLisp 3.2.

commit c527b622 (2020-04-29): Version 3.2 - 29th April 2020

Fixes #29
closed 2020-04-29 by technoblogy