technoblogy/ulisp-esp #33: I need some guide to run uLisp on esp32

bskaplou, 2020-06-24:
Copipasted https://github.com/technoblogy/ulisp-esp/blob/master/ulisp-esp.ino into sketch and getting gcc errors.. 
Any ideas how to fix?

```
Library/Arduino15/packages/esp32/tools/xtensa-esp32-elf-gcc/1.22.0-80-g6c4433a-5.2.0/bin/xtensa-esp32-elf-g++ -DESP_PLATFORM "-DMBEDTLS_CONFIG_FILE=\"mbedtls/esp_config.h\"" -DHAVE_CONFIG_H -DGCC_NOT_5_2_0=0 -DWITH_POSIX -I/Users/bskaplou/Library/Arduino15/packages/esp32/hardware/esp32/1.0.4/tools/sdk/include/config -I/Users/bskaplou/Library/Arduino15/packages/esp32/hardware/esp32/1.0.4/tools/sdk/include/app_trace -I/Users/bskaplou/Library/Arduino15/packages/esp32/hardware/esp32/1.0.4/tools/sdk/include/app_update -I/Users/bskaplou/Library/Arduino15/packages/esp32/hardware/esp32/1.0.4/tools/sdk/include/asio -I/Users/bskaplou/Library/Arduino15/packages/esp32/hardware/esp32/1.0.4/tools/sdk/include/bootloader_support -I/Users/bskaplou/Library/Arduino15/packages/esp32/hardware/esp32/1.0.4/tools/sdk/include/bt -I/Users/bskaplou/Library/Arduino15/packages/esp32/hardware/esp32/1.0.4/tools/sdk/include/coap -I/Users/bskaplou/Library/Arduino15/packages/esp32/hardware/esp32/1.0.4/tools/sdk/include/console -I/Users/bskaplou/Library/Arduino15/packages/esp32/hardware/esp32/1.0.4/tools/sdk/include/driver -I/Users/bskaplou/Library/Arduino15/packages/esp32/hardware/esp32/1.0.4/tools/sdk/include/esp-tls -I/Users/bskaplou/Library/Arduino15/packages/esp32/hardware/esp32/1.0.4/tools/sdk/include/esp32 -I/Users/bskaplou/Library/Arduino15/packages/esp32/hardware/esp32/1.0.4/tools/sdk/include/esp_adc_cal -I/Users/bskaplou/Library/Arduino15/packages/esp32/hardware/esp32/1.0.4/tools/sdk/include/esp_event -I/Users/bskaplou/Library/Arduino15/packages/esp32/hardware/esp32/1.0.4/tools/sdk/include/esp_http_client -I/Users/bskaplou/Library/Arduino15/packages/esp32/hardware/esp32/1.0.4/tools/sdk/include/esp_http_server -I/Users/bskaplou/Library/Arduino15/packages/esp32/hardware/esp32/1.0.4/tools/sdk/include/esp_https_ota -I/Users/bskaplou/Library/Arduino15/packages/esp32/hardware/esp32/1.0.4/tools/sdk/include/esp_ringbuf -I/Users/bskaplou/Library/Arduino15/packages/esp32/hardware/esp32/1.0.4/tools/sdk/include/ethernet -I/Users/bskaplou/Library/Arduino15/packages/esp32/hardware/esp32/1.0.4/tools/sdk/include/expat -I/Users/bskaplou/Library/Arduino15/packages/esp32/hardware/esp32/1.0.4/tools/sdk/include/fatfs -I/Users/bskaplou/Library/Arduino15/packages/esp32/hardware/esp32/1.0.4/tools/sdk/include/freemodbus -I/Users/bskaplou/Library/Arduino15/packages/esp32/hardware/esp32/1.0.4/tools/sdk/include/freertos -I/Users/bskaplou/Library/Arduino15/packages/esp32/hardware/esp32/1.0.4/tools/sdk/include/heap -I/Users/bskaplou/Library/Arduino15/packages/esp32/hardware/esp32/1.0.4/tools/sdk/include/idf_test -I/Users/bskaplou/Library/Arduino15/packages/esp32/hardware/esp32/1.0.4/tools/sdk/include/jsmn -I/Users/bskaplou/Library/Arduino15/packages/esp32/hardware/esp32/1.0.4/tools/sdk/include/json -I/Users/bskaplou/Library/Arduino15/packages/esp32/hardware/esp32/1.0.4/tools/sdk/include/libsodium -I/Users/bskaplou/Library/Arduino15/packages/esp32/hardware/esp32/1.0.4/tools/sdk/include/log -I/Users/bskaplou/Library/Arduino15/packages/esp32/hardware/esp32/1.0.4/tools/sdk/include/lwip -I/Users/bskaplou/Library/Arduino15/packages/esp32/hardware/esp32/1.0.4/tools/sdk/include/mbedtls -I/Users/bskaplou/Library/Arduino15/packages/esp32/hardware/esp32/1.0.4/tools/sdk/include/mdns -I/Users/bskaplou/Library/Arduino15/packages/esp32/hardware/esp32/1.0.4/tools/sdk/include/micro-ecc -I/Users/bskaplou/Library/Arduino15/packages/esp32/hardware/esp32/1.0.4/tools/sdk/include/mqtt -I/Users/bskaplou/Library/Arduino15/packages/esp32/hardware/esp32/1.0.4/tools/sdk/include/newlib -I/Users/bskaplou/Library/Arduino15/packages/esp32/hardware/esp32/1.0.4/tools/sdk/include/nghttp -I/Users/bskaplou/Library/Arduino15/packages/esp32/hardware/esp32/1.0.4/tools/sdk/include/nvs_flash -I/Users/bskaplou/Library/Arduino15/packages/esp32/hardware/esp32/1.0.4/tools/sdk/include/openssl -I/Users/bskaplou/Library/Arduino15/packages/esp32/hardware/esp32/1.0.4/tools/sdk/include/protobuf-c -I/Users/bskaplou/Library/Arduino15/packages/esp32/hardware/esp32/1.0.4/tools/sdk/include/protocomm -I/Users/bskaplou/Library/Arduino15/packages/esp32/hardware/esp32/1.0.4/tools/sdk/include/pthread -I/Users/bskaplou/Library/Arduino15/packages/esp32/hardware/esp32/1.0.4/tools/sdk/include/sdmmc -I/Users/bskaplou/Library/Arduino15/packages/esp32/hardware/esp32/1.0.4/tools/sdk/include/smartconfig_ack -I/Users/bskaplou/Library/Arduino15/packages/esp32/hardware/esp32/1.0.4/tools/sdk/include/soc -I/Users/bskaplou/Library/Arduino15/packages/esp32/hardware/esp32/1.0.4/tools/sdk/include/spi_flash -I/Users/bskaplou/Library/Arduino15/packages/esp32/hardware/esp32/1.0.4/tools/sdk/include/spiffs -I/Users/bskaplou/Library/Arduino15/packages/esp32/hardware/esp32/1.0.4/tools/sdk/include/tcp_transport -I/Users/bskaplou/Library/Arduino15/packages/esp32/hardware/esp32/1.0.4/tools/sdk/include/tcpip_adapter -I/Users/bskaplou/Library/Arduino15/packages/esp32/hardware/esp32/1.0.4/tools/sdk/include/ulp -I/Users/bskaplou/Library/Arduino15/packages/esp32/hardware/esp32/1.0.4/tools/sdk/include/vfs -I/Users/bskaplou/Library/Arduino15/packages/esp32/hardware/esp32/1.0.4/tools/sdk/include/wear_levelling -I/Users/bskaplou/Library/Arduino15/packages/esp32/hardware/esp32/1.0.4/tools/sdk/include/wifi_provisioning -I/Users/bskaplou/Library/Arduino15/packages/esp32/hardware/esp32/1.0.4/tools/sdk/include/wpa_supplicant -I/Users/bskaplou/Library/Arduino15/packages/esp32/hardware/esp32/1.0.4/tools/sdk/include/xtensa-debug-module -I/Users/bskaplou/Library/Arduino15/packages/esp32/hardware/esp32/1.0.4/tools/sdk/include/esp-face -I/Users/bskaplou/Library/Arduino15/packages/esp32/hardware/esp32/1.0.4/tools/sdk/include/esp32-camera -I/Users/bskaplou/Library/Arduino15/packages/esp32/hardware/esp32/1.0.4/tools/sdk/include/esp-face -I/Users/bskaplou/Library/Arduino15/packages/esp32/hardware/esp32/1.0.4/tools/sdk/include/fb_gfx -std=gnu++11 -Os -g3 -Wpointer-arith -fexceptions -fstack-protector -ffunction-sections -fdata-sections -fstrict-volatile-bitfields -mlongcalls -nostdlib -w -Wno-error=maybe-uninitialized -Wno-error=unused-function -Wno-error=unused-but-set-variable -Wno-error=unused-variable -Wno-error=deprecated-declarations -Wno-unused-parameter -Wno-unused-but-set-parameter -Wno-missing-field-initializers -Wno-sign-compare -fno-rtti -MMD -c -DF_CPU=240000000L -DARDUINO=10812 -DARDUINO_ESP32_DEV -DARDUINO_ARCH_ESP32 "-DARDUINO_BOARD=\"ESP32_DEV\"" "-DARDUINO_VARIANT=\"esp32\"" -DESP32 -DCORE_DEBUG_LEVEL=0 -DBOARD_HAS_PSRAM -mfix-esp32-psram-cache-issue -I/Users/bskaplou/Library/Arduino15/packages/esp32/hardware/esp32/1.0.4/cores/esp32 -I/Users/bskaplou/Library/Arduino15/packages/esp32/hardware/esp32/1.0.4/variants/esp32 -I/Users/bskaplou/Library/Arduino15/packages/esp32/hardware/esp32/1.0.4/libraries/BLE/src -I/Users/bskaplou/Library/Arduino15/packages/esp32/hardware/esp32/1.0.4/libraries/SPI/src -I/Users/bskaplou/Library/Arduino15/packages/esp32/hardware/esp32/1.0.4/libraries/Wire/src -I/Users/bskaplou/Library/Arduino15/packages/esp32/hardware/esp32/1.0.4/libraries/EEPROM/src -I/Users/bskaplou/Library/Arduino15/packages/esp32/hardware/esp32/1.0.4/libraries/WiFi/src /var/folders/41/0vvh_5k56xs75cqn9ygdbxmc0000gn/T/arduino_build_849625/sketch/ESP32_BLE_client.ino.cpp -o /var/folders/41/0vvh_5k56xs75cqn9ygdbxmc0000gn/T/arduino_build_849625/sketch/ESP32_BLE_client.ino.cpp.o
ulisp:221:15: error: variable or field 'errorsub' declared void
 void errorsub(symbol_t fname, PGM_P string) {
               ^
ulisp:221:15: error: 'symbol_t' was not declared in this scope
ulisp:221:31: error: expected primary-expression before 'const'
 void errorsub(symbol_t fname, PGM_P string) {
                               ^
ulisp:231:12: error: variable or field 'error' declared void
 void error (symbol_t fname, PGM_P string, object *symbol) {
            ^
```

technoblogy, 2020-06-24:
Not sure why you're getting that error. The definition of symbol_t is in the file:

    typedef unsigned int symbol_t;

What platform are you compiling on? I've just tried compiling uLisp ESP Version 3.3 on the Arduino IDE for the Adafruit ESP 32 Feather board, and it compiles fine without errors.

bskaplou, 2020-06-24:
Sorry my bad.... started from empty sketch and no errors...
Lisping on ;)

closed 2020-06-24 by bskaplou

technoblogy, 2020-06-24:
No problem. Have fun with uLisp!