technoblogy/ulisp-esp #54: Use as standalone library?

dragoncoder047, 2022-07-16:
I'm planning on using Microlisp as a scripting language for a robot I am building, and I want to automatically run code on a microSD card upon startup instead of launching the REPL, so that I don't need to connect to it over serial to do anything. (I will be using [this](https://www.sparkfun.com/products/18018) ESP32 board if that helps.)

From what I have read in the documentation, I need to add code to the `LispLibrary` at the top, uncomment the `#define`s to enable SD card support, add in my C functions at the bottom, and then what do I need to change in the `setup()` and `loop()` to get it out of REPL mode?

I don't want to use the `(load-image)` support because it would become corrupted if I edit the C code and I would have to tediously load all the Lisp all over again and then `(save-image 'main)` or something like that every time. I just want to be able to flash the firmware onto the ESP32, plug in a microSD card with a bunch of `.lisp` files on it, and have it run itself.

I know I could have posted this in the forums, but I really don't want to have to create a brand-new password-protected account just for one stupid question.

If this is already in the documentation, please point me to the appropriate page. My apologies in advance if this is the case.

dragoncoder047, 2022-07-16:
Also, unrelated question, how can I split the `ulisp-esp.ino` up so I can include my "extra" functions in a separate file? Something like this:

```cpp
#include "microlisp/core-esp.cpp"

object *fn_myfun (object *args, object *env) {
  // do something
  return nil;
}
// how to register it in the lookup table??

void setup() {
  // other setup as necessary
  microlisp_init();
}

void loop() {
  // NOW WHAT?
}
```

I have seen on the forums that the source file isn't quite correct C++ (missing forward references, etc), so what changes would need to be made there (as well as the problems pointed out in the comments on my snippet above)?

technoblogy, 2022-07-17:
> I know I could have posted this in the forums, but I really don't want to have to create a brand-new password-protected account just for one stupid question.

Thanks for the issue. These are two very good questions, and I've actually been thinking about the second one over the last few days. Could you put them into two separate issues, so I can answer them in separate threads? Or, even better, post them onto the uLisp forum, because the questions apply to all versions of uLisp, and I think other uLisp users would be interested in the answers.

> I'm planning on using Microlisp as a scripting language for a robot I am building

I assume you mean uLisp! Thanks, David

dragoncoder047, 2022-07-17:
> Could you put them into two separate issues, so I can answer them in separate threads? Or, even better, post them onto the uLisp forum, because the questions apply to all versions of uLisp, and I think other uLisp users would be interested in the answers.

Again, I really don't feel the need to create a whole nother account just for this. I'm going to file them as issues on the main uLisp repository.

> I assume you mean uLisp!

Is is <strong>u</strong>Lisp or <strong>&micro;</strong>Lisp? Or is either acceptable?

technoblogy, 2022-07-17:
> Is is uLisp or µLisp? Or is either acceptable?

uLisp (pronounced "you-lisp"). The Greek mu symbol doesn't look good in many typefaces, and causes problems with URLs etc.

dragoncoder047, 2022-07-17:
>uLisp (pronounced "you-lisp"). 

Thanks for correcting me!

>The Greek mu symbol doesn't look good in many typefaces, and causes problems with URLs etc.

I'm just wondering- was it originally supposed to be "micro-Lisp" and you changed it because of that?

technoblogy, 2022-07-17:
Yes, it means "micro Lisp", but there was already a legacy product with that name, so I went with uLisp.

dragoncoder047, 2022-08-15:
Closing because I moved it to the main uLisp repo already.

closed 2022-08-15 by dragoncoder047