technoblogy/ulisp-esp #55: Assembler for  ESP32/ESP8266?

dragoncoder047, 2022-08-15:
I've noticed that there are assemblers for AVR, ARM, and RISC-V versions of uLisp, but not ESP. Why? Is it not possible with the ESP architecture?

I'm wondering because I would like to use the assembler to drive some WS2182's (NeoPixels) but there is no version for ESP32. I would rather not use a C library because I am trying to add some other flash-hungry libraries to uLisp and I want as much space as possible.

technoblogy, 2022-08-15:
The uLisp assemblers generate machine code in RAM and then execute it, so they rely on the ability to execute code in RAM. I did some tests on the ESP32 and got stuck at this point, so I didn't go any further. See:

https://www.esp32.com/viewtopic.php?f=19&t=22217

If you can see any way around this then we could make some progress. Thanks! David

dragoncoder047, 2022-08-15:
That's what I was afraid of... It may not be possible on ESP32s. I don't know enough about ESP32's to be able to help you, sorry. I guess that I will have to use the C++ library.

The good thing is that I figured out how to change the partition scheme for my ESP32's internal flash and now I have 3MB (29% used by uLisp) as opposed to the default only 1.2MB (64% used by uLisp).

closed 2022-08-15 by dragoncoder047