technoblogy/ulisp-esp #57: with-sd-card panics if filename is not provided

dragoncoder047, 2022-09-24:
I downloaded "Commented uLisp 4.3 for ESP" from [here](http://www.ulisp.com/list?43UC).

First, compiler said on lines 724 and 793 `error: too few arguments to function 'char* MakeFilename(object*, char*)'` (inside `loadimage()` and `saveimage()`) so I added a `char buffer[BUFFERSIZE]` which seemed to fix it.

Tried a few tests:

```lisp
(with-sd-card (s"foo.txt")s)
; panics
; Guru Meditation Error: Core  1 panic'ed (LoadProhibited). Exception was unhandled.
(with-sd-card (s "foo.txt") s)
; works as expected
'(with-sd-card (s"foo.txt")s)
; prints (with-sd-card (s\"foo.txt\") s)
(with-sd-card (foo) s)
; panics
(with-sd-card foo s)
; panics
```

Backtrace points to the second of these lines in `sp_withsdcard`:

```cpp
object *params = first(args);
object *var = first(params); // <<<<<<<<<< Dereferenced a null pointer here
```

does not check if `params` is null.

seems to be similar to #52.

any idea for a fix?

technoblogy, 2022-09-25:
Thanks for finding this! This has revealed several bugs which will take a bit of time to fix, and I'll release a new version in due course. In the meantime please include spaces as in:

    (with-sd-card (s "foo.txt") s)

technoblogy, 2022-09-25:
What IDE are you using that gives useful backtraces like this?

    object *params = first(args);
    object *var = first(params); // <<<<<<<<<< Dereferenced a null pointer here

Thanks!

dragoncoder047, 2022-09-25:
> What IDE are you using that gives useful backtraces like this?

https://github.com/me-no-dev/EspExceptionDecoder

That's how I was also able to track #56 down to that line in `markobject()` too.

technoblogy, 2022-09-25:
Fixing this properly revealed some other bugs, so I'm very grateful for the feedback! There's a new version that should fix all the bugs here:

[ESP Version 4.3a](http://www.ulisp.com/list?43UC)

This detects the missing parameter for **with-sd-card**, but is also more tolerant of missing spaces where they shouldn't be needed, such as in:

    (with-sd-card (s"foo.txt")s)

It also fixes a problem with reading in bit arrays. Let me know how you get on with it!

I'll update the other platforms soon.

dragoncoder047, 2022-09-25:
Great, it works now. That version isn't pushed to GitHub?

technoblogy, 2022-09-25:
> That version isn't pushed to GitHub?

It is now!

commit a46baae5 (2022-09-25): Version 4.3a - 25th September 2022

Fixes #57
closed 2022-09-25 by technoblogy