technoblogy/ulisp-esp #59: error in docstring

dragoncoder047, 2023-03-26:
The docstring for `log` doesn't include `log` (it just starts with `(number [base])`)

Should be `(log number [base])`.

dragoncoder047, 2023-03-26:
The docstrings for `truncate` and `round` also have the docstring of `floatp` by mistake. What should they be?

dragoncoder047, 2023-03-26:
* `lognot` has the entire docstring of `prin1-to-string`

technoblogy, 2023-03-26:
Thanks for spotting these. Will fix.

> The docstrings for truncate and round also have the docstring of floatp by mistake. What should they be?

    "(truncate number [divisor])\n"
    "Returns the integer part of number/divisor. If divisor is omitted it defaults to 1."

    "(round number [divisor])\n"
    "Returns the integer closest to number/divisor. If divisor is omitted it defaults to 1."

(from [Floating-point extensions](http://www.ulisp.com/show?1G5Y)).

technoblogy, 2023-03-26:
PS When did this emoji menu appear in GitHub issues? I kept thinking I'd typed in a smiley by mistake.

technoblogy, 2023-04-01:
Fixed in 4.4b - thanks!

closed 2023-04-01 by technoblogy