technoblogy/ulisp-esp #63: Any suggestions for how to solve compiler warnings in uLisp 4.4?

technoblogy, 2023-03-31:
The new error-handling functions in uLisp 4.4, **unwind-protect**, **ignore-errors**, and **error** work correctly, but cause compiler warnings on all platforms:
````text
/Users/david/Documents/Arduino/LispESP198/LispESP198.ino: In function 'object* sp_unwindprotect(object*, object*)':
/Users/david/Documents/Arduino/LispESP198/LispESP198.ino:4031:11: warning: variable 'result' might be clobbered by 'longjmp' or 'vfork' [-Wclobbered]
   object *result;
           ^~~~~~
/Users/david/Documents/Arduino/LispESP198/LispESP198.ino: In function 'object* sp_ignoreerrors(object*, object*)':
/Users/david/Documents/Arduino/LispESP198/LispESP198.ino:4063:8: warning: variable 'signaled' might be clobbered by 'longjmp' or 'vfork' [-Wclobbered]
   bool signaled = false;
        ^~~~~~~~
/Users/david/Documents/Arduino/LispESP198/LispESP198.ino:4054:34: warning: argument 'args' might be clobbered by 'longjmp' or 'vfork' [-Wclobbered]
 object *sp_ignoreerrors (object *args, object *env) {
                          ~~~~~~~~^~~~
````
Can anyone explain what is causing these, and suggest how to eliminate them?

dragoncoder047, 2023-05-27:
I compiled uLisp again today on a fresh install of arduino & esp core and also got this warning as well as the longjmp warnings:

```
/home/dc047/Desktop/arduino stuff/ulisp-esp32/ulisp.hpp: In function 'char fromradix40(char)':
/home/dc047/Desktop/arduino stuff/ulisp-esp32/ulisp.hpp:774:11: warning: comparison is always false due to limited range of data type [-Wtype-limits]
     if (n < 0 || n >= 40) return 0;
         ~~^~~
```

Weird... `n` is a `char`, which is signed AFAIK...

technoblogy, 2023-05-28:
Actually `char` can be signed or unsigned:

https://stackoverflow.com/questions/2054939/is-char-signed-or-unsigned-by-default

I should probably fix that so the warning cannot appear whatever the definition of `char` - thanks!

dragoncoder047, 2023-05-28:
> Actually `char` can be signed or unsigned.

OK, good to know that `char` $\equiv$ `byte` on ESP32 at least.

technoblogy, 2026-02-24:
This was fixed in 4.8g.