technoblogy/ulisp-esp #65: very simple keyword support

dragoncoder047, 2023-04-03:
earlier I think I mentioned that I implemented generic keywords by automatically wrapping them in `(quote)` in the reader, but that's probably a bad solution if I ever implement macros. I just came up with another solution that is super simple and works with macros:

```diff
    // in eval ()
    if (symbolp(form)) {
+       if (nthchar(princtostring(form), 0) == ':') return form; // Keyword
        symbol_t name = form->name;
        object* pair = value(name, env);
        if (pair != NULL) return cdr(pair);
        pair = value(name, GlobalEnv);
        if (pair != NULL) return cdr(pair);
        else if (builtinp(name)) return form;
        Context = NIL;
        error(PSTR("undefined"), form);
    }
```

technoblogy, 2023-04-04:
Neat!

dragoncoder047, 2023-04-04:
As a side note I still think you should keep the existing keyword mechanism (entry in lookup table) because then it allows you to associate meaning to a particular keyword in a particular context, such as `:input`/`:output`/`:input-pullup`, and avoids the need for an `if`-statement cascade or a secondary lookup table because the "value" of the function is the particular number for that context (`INPUT`/`OUTPUT`/`INPUT_PULLUP` etc.)

technoblogy, 2023-04-04:
Agreed - and I don't think there's currently any other useful function for keywords.

dragoncoder047, 2023-04-04:
> I don't think there's currently any other useful function for keywords.

Well, you could do `&key` arguments. The algorithm I see would be to take the `:foo foo-value` pairs passed in, and match the keyword to the parameter name, and then pop the parameter name from the definition because you can only have it once. Then everything else becomes the `&rest` arguments (or an error `no such keyword argument: :foo` if there is no `&rest`).