technoblogy/ulisp-esp #66: with-sd-card gives incorrect error message on long symbol as filename, panics on short symbol

dragoncoder047, 2023-04-05:
```
> (with-sd-card (foo 'barbazbingbam) foo)
Error: 'quote' problem reading from SD card or invalid filename: barbazbingbam
> (with-sd-card (foo 'bar) foo)
Guru Meditation Error: Core  1 panic'ed (LoadProhibited). Exception was unhandled.
```

The first one probably just involves saving Context when entering sp_withsdcard, and setting it just before throwing an error.

The second one, however, looks to be a little more complex. The backtrace points to it passing the symbol to MakeFilename, which then tries to call nthchar on it and it crashes.

Probably what it should do is check to make sure the second parameter is a string. Or maybe MakeFilename should do that.

technoblogy, 2023-04-05:
Thanks - I'll investigate.

technoblogy, 2023-04-05:
You're correct about the first one; the context should be saved around the **eval()** in **sp_withsdcard()**;

In the second one, **filename** should be checked to make sure it's a string before calling **MakeFilename()**, as it is in **saveimage()** and **loadimage()**.

Thanks for finding these.

commit f22e87e8 (2023-06-30): Release 4.4d - 30th June 2023

Fixes #66, fixes #67, fixes #76
closed 2023-06-30 by technoblogy