technoblogy/ulisp-esp #67: possible bug: `(read-line)` processes backslash escapes (it uses readstring())

dragoncoder047, 2023-04-06:
What would happen if you tried to `(read-line)` on a stream that contained backslashes?

Common Lisp does not do this:

```lisp
CL-USER> (defvar s "foo \\
bar")
"foo \\
bar"
CL-USER> (defvar st (make-string-input-stream s))
#<INPUT STRING-INPUT-STREAM>
CL-USER> (princ (read-line st))
foo \
"foo \\"
```

I can't test it with uLisp because it doesn't have (make-string-input-stream), but I don't think it would stop at the newline if a backslash immediately follows it.

I think the intention of `(read-line)` was to ignore backslashes and include them.

technoblogy, 2023-04-07:
Good point - thanks.

dragoncoder047, 2023-04-07:
I think an easy solution would be to have a third parameter `doEscape` and pass true in `(read)` and false in `(read-line)`.

technoblogy, 2023-04-07:
Yes, I was thinking along the same lines.

dragoncoder047, 2023-04-07:
It just dawned on me that you could implement something like `(make-string-input-stream)` / `(with-input-from-string)` by simply using `gstr()`. Unfortunately until #64 is resolved it will only work for one stringstream, but that's already an inherent limitation with the current implementation.

technoblogy, 2023-04-12:
Good point. I've already got **with-output-to-string** so I don't know why I didn't include **with-input-from-string**.

commit f22e87e8 (2023-06-30): Release 4.4d - 30th June 2023

Fixes #66, fixes #67, fixes #76
closed 2023-06-30 by technoblogy