technoblogy/ulisp-esp #68: websockets?

dragoncoder047, 2023-04-07:
I found an Arduino websockets library that works with the existing wifi library: https://github.com/gilmaimon/ArduinoWebsockets

It seems simple enough to implement similar websocket functions as the existing wifi functions in uLisp, as the interface is so similar.

As an implementation note, it does look like `server.available()` doesn't get you a client, it instead returns if the websocket server is running. `server.accept()` actually gets the client, but will block if there is no client, `server.poll()` returns true if there is a waiting client.

As for the client end, because the design of websockets is to push data without being requested, not respond to a request as regular HTTP is, actually getting data from the websocket using the callback will probably involve some sort of queue to pipe the data from the client callback to the `gfun_t` of the `websocket-stream`.

technoblogy, 2023-04-07:
Perhaps write it as an Extensions file, using the new facility introduced in uLisp release 4.4?

dragoncoder047, 2023-04-07:
> Perhaps write it as an Extensions file, using the new facility introduced in uLisp release 4.4?

I could, but it involves adding a new stream so it would involve editing the `streams` enums, `gstreamfun()` and `pstreamfun()`.

technoblogy, 2023-04-08:
True. What are the benefits and potential applications for such a feature?

dragoncoder047, 2023-04-08:
> What are the benefits and potential applications for such a feature?

I think the biggest benefit is that once the socket connection is established, the microcontroller can push data to all of the web pages that have established websocket connections, and they don't have to poll repeatedly and overload the microcontroller with having to parse a gazillion HTTP requests.