technoblogy/ulisp-esp #73: using esp32 sleep function

dragoncoder047, 2023-04-27:
Right now the `(sleep)` function doesn't actually put the CPU to sleep, it just burns time in a delay() loop.

However, the ESP32 does have a light-sleep function that preserves the state - care to try it out?

```cpp
esp_sleep_enable_timer_wakeup(secs * 1000000UL);
esp_light_sleep_start();
```

Also, I might add that maybe you can allow `(sleep)` to be called with no arguments, which means indefinite sleep -- on esp32 it just directly calls `esp_light_sleep_start()`, on AVR it calls `sleep_enable(); sleep_cpu();` without enabling the watchdog for interrupts, etc.

technoblogy, 2023-04-27:
Yes, the **sleep** function puts the CPU to sleep on AVR and some ARM platforms, but I couldn't work out how to do it on the ESP32.

Also, I have done what you suggested on AVR - (sleep) goes to sleep for ever, and you can arrange an interrupt to wake you up.

I'll try implementing it on ESP32 as you've suggested.

technoblogy, 2023-06-30:
Haven't done this yet - sorry!