technoblogy/ulisp-esp #77: Latest ESP32 Arduino core gives less available RAM for uLisp on ESP32-S2

technoblogy, 2023-06-30:
I tested the latest release of uLisp on an ESP32-S2 board (Adafruit ESP32-S2 Feather) with the current Arduino core, 2.0.9, and found I have over 8Kbytes less available RAM than under the previous core I tested, 2.0.5.

This doesn't seem to happen with ESP32 boards (eg Adafruit ESP32 Feather).

In ESP uLisp release 4.4d I've therefore reduced the amount of Lisp workspace on the ESP32-S2 boards  from 9216 objects to 8160 objects. Otherwise compilation fails with the error:

    error: ld returned 1 exit status

I've tried to track down what's causing this, but without success. I suspect it may be the Wi-Fi libraries.

Can anyone can shed any light on this?

dragoncoder047, 2023-06-30:
I think it may be the *USB* interface, not Wifi, because the S2 does USB-to-serial conversion by itself, whereas the regular ESP32 has to use a external converter.

technoblogy, 2023-06-30:
Interesting suggestion. Actually, there is an ESP32-S2 board with an external converter that would enable me to test this theory:

http://www.ulisp.com/show?3TQF#esp32-s2-saola-1

technoblogy, 2023-06-30:
You're right! The Espressif ESP32-S2 Dev board includes a CP2102N USB to Serial converter used by default, or you can choose the native USB CDC.

With the hardware USB converter I can get 9216 objects of workspace, whereas with the native USB CDC I can only get 8160 objects, over 8Kbytes less.

However, so far I haven't been able to make a minimal reproducible example to demonstrate it.

closed 2024-07-26 by technoblogy