technoblogy/ulisp-esp #78: Remappable peripheral pins

hasn0life, 2023-08-31:
Hello and thank you for making this wonderful language.

I was wondering if it would be possible to allow the I2Cs and UARTs to do pin remapping on the ESP32 since it supports it. I don't know what a good interface would look like tho, would it be another function or more arguments or optional arguments? Right now I'm modifying the codebase to do it, but maybe that's the best way?

hasn0life, 2023-09-01:
I'm not sure because you'd have to specify the pins every time, which is annoying on a microcontroller if you forget or map them wrong. Also if you're moving code or pins you'll have to add it in all the places if you didn't initially use variables for the pins. I'm leaning towards a function to set them once at the beginning, perhaps something like
```
(defun seti2cpins (sda scl [port])
      ...)
```

crackwitz, 2023-09-01:
Would it be too unreasonable to stick as closely as possible to esp-idf APIs?

dragoncoder047, 2023-09-01:
> Would it be too unreasonable to stick as closely as possible to esp-idf APIs?

Well, uLisp is platform-independent, keep in mind that there are versions of it for AVR, ARM, and RISC-V processors as well as ESPs. The SDK's are all different for each processor, so David has tried to simplify so it's the same from this Lisp side on all platforms.

crackwitz, 2023-09-01:
My bad. The name made me think this was using espressif APIs. I didn't realize this was actually based on Arduino.