technoblogy/ulisp-esp #82: Compiling failure on Adafruit QT PY ESP32-S2

fighne, 2024-07-16:
Previous version (4.4d)  compiles current version fails.
Set compiler output to verbose and all messages in attached text file
[error.txt](https://github.com/user-attachments/files/16241579/error.txt)

dragoncoder047, 2024-07-16:
Yikes, that's an error from the Arduino build program, not the compiler. I have no idea what could cause it. What Arduino IDE version and OS are you on?

technoblogy, 2024-07-16:
When you successfully compiled uLisp 4.4d did you use the same ESP32 Arduino Core, 3.0.2?

I've had problems with the latest ESP32 cores; try an older core; eg 2.0.12.

fighne, 2024-07-16:
ESP32 Arduino Core is currently 3.0.2, will downgrade and try to recompile.

fighne, 2024-07-16:
With core set at 2.0.16 
4.4d compiles fine 
4.6 errors with the following
Arduino: 1.8.19 (Linux), Board: "Adafruit QT Py ESP32-S2, Enabled, Disabled, Disabled, Internal USB, Enabled, Default 4MB with spiffs (1.2MB APP/1.5MB SPIFFS), 240MHz (WiFi), QIO, 80MHz, 4MB (32Mb), 921600, None, Disabled"

WARNING: Category '' in library ESP Insights is not valid. Setting to 'Uncategorized'
WARNING: Category '' in library ESP RainMaker is not valid. Setting to 'Uncategorized'
WARNING: Category '' in library WiFiProv is not valid. Setting to 'Uncategorized'
/home/fighne/Documents/GitHub/ulisp/ulisp-esp-4v6/ulisp-esp/ulisp-esp.ino:83: warning: "LED_BUILTIN" redefined
   #define LED_BUILTIN 13
 
In file included from /home/fighne/.arduino15/packages/esp32/hardware/esp32/2.0.16/cores/esp32/esp32-hal-gpio.h:29,
                 from /home/fighne/.arduino15/packages/esp32/hardware/esp32/2.0.16/cores/esp32/esp32-hal.h:83,
                 from /home/fighne/.arduino15/packages/esp32/hardware/esp32/2.0.16/cores/esp32/Arduino.h:36,
                 from sketch/ulisp-esp.ino.cpp:1:
/home/fighne/.arduino15/packages/esp32/hardware/esp32/2.0.16/variants/adafruit_qtpy_esp32s2/pins_arduino.h:17: note: this is the location of the previous definition
 #define LED_BUILTIN LED_BUILTIN  // allow testing #ifdef LED_BUILTIN
 
/home/fighne/Documents/GitHub/ulisp/ulisp-esp-4v6/ulisp-esp/ulisp-esp.ino: In function 'object* eval(object*, object*)':
/home/fighne/Documents/GitHub/ulisp/ulisp-esp-4v6/ulisp-esp/ulisp-esp.ino:5756:24: warning: unused variable 'start' [-Wunused-variable]
   static unsigned long start = 0;
                        ^~~~~
/home/fighne/Documents/GitHub/ulisp/ulisp-esp-4v6/ulisp-esp/ulisp-extensions.ino: At global scope:
ulisp-extensions:41:21: error: redefinition of 'tbl_entry_t* tables []'
 tbl_entry_t *tables[] = {lookup_table, lookup_table2};
                     ^
/home/fighne/Documents/GitHub/ulisp/ulisp-esp-4v6/ulisp-esp/ulisp-esp.ino:5678:14: note: 'tbl_entry_t* tables [2]' previously defined here
 tbl_entry_t *tables[] = {lookup_table, NULL};
              ^~~~~~
ulisp-extensions:42:31: error: redefinition of 'const unsigned int tablesizes []'
 const unsigned int tablesizes[] = { arraysize(lookup_table), arraysize(lookup_table2) };
                               ^
/home/fighne/Documents/GitHub/ulisp/ulisp-esp-4v6/ulisp-esp/ulisp-esp.ino:5679:20: note: 'const unsigned int tablesizes [2]' previously defined here
 const unsigned int tablesizes[] = { arraysize(lookup_table), 0 };
                    ^~~~~~~~~~
ulisp-extensions:44:20: error: redefinition of 'tbl_entry_t* table(int)'
 const tbl_entry_t *table (int n) {
                    ^~~~~
/home/fighne/Documents/GitHub/ulisp/ulisp-esp-4v6/ulisp-esp/ulisp-esp.ino:5681:20: note: 'tbl_entry_t* table(int)' previously defined here
 const tbl_entry_t *table (int n) {
                    ^~~~~
ulisp-extensions:48:14: error: redefinition of 'unsigned int tablesize(int)'
 unsigned int tablesize (int n) {
              ^~~~~~~~~
/home/fighne/Documents/GitHub/ulisp/ulisp-esp-4v6/ulisp-esp/ulisp-esp.ino:5685:14: note: 'unsigned int tablesize(int)' previously defined here
 unsigned int tablesize (int n) {
              ^~~~~~~~~
/home/fighne/Documents/GitHub/ulisp/ulisp-esp-4v6/ulisp-esp/ulisp-extensions.ino:44:20: warning: 'tbl_entry_t* table(int)' defined but not used [-Wunused-function]
 const tbl_entry_t *table (int n) {
                    ^~~~~
/home/fighne/Documents/GitHub/ulisp/ulisp-esp-4v6/ulisp-esp/ulisp-esp.ino:5756:24: warning: 'start' defined but not used [-Wunused-variable]
   static unsigned long start = 0;
                        ^~~~~
exit status 1
redefinition of 'tbl_entry_t* tables []'

This report would have more information with
"Show verbose output during compilation"
option enabled in File -> Preferences.

dragoncoder047, 2024-07-16:
Most of those "redefinition" errors are related to the extensions; see [here](http://forum.ulisp.com/t/adafruit-qt-py-esp32-s2-compiling/1419/2?u=dragoncoder047) for the fix. It's literally deleting two characters to fix, super easy to miss, and super confusing when you do. There's not much anyone can do to change the error messages unfortunately ;)

The first "redefinition" error (of LED_BUILTIN) I'm not sure where it is coming from and it's unrelated to the above fix...

fighne, 2024-07-16:
Ah! thank you well spotted and corrected in the src code
New error
Arduino: 1.8.19 (Linux), Board: "Adafruit QT Py ESP32-S2, Enabled, Disabled, Disabled, Internal USB, Enabled, Default 4MB with spiffs (1.2MB APP/1.5MB SPIFFS), 240MHz (WiFi), QIO, 80MHz, 4MB (32Mb), 921600, None, Disabled"

WARNING: Category '' in library ESP Insights is not valid. Setting to 'Uncategorized'
WARNING: Category '' in library ESP RainMaker is not valid. Setting to 'Uncategorized'
WARNING: Category '' in library WiFiProv is not valid. Setting to 'Uncategorized'
/home/fighne/Documents/GitHub/ulisp/ulisp-esp-4v6/ulisp-esp/ulisp-esp.ino:83: warning: "LED_BUILTIN" redefined
   #define LED_BUILTIN 13
 
In file included from /home/fighne/.arduino15/packages/esp32/hardware/esp32/2.0.16/cores/esp32/esp32-hal-gpio.h:29,
                 from /home/fighne/.arduino15/packages/esp32/hardware/esp32/2.0.16/cores/esp32/esp32-hal.h:83,
                 from /home/fighne/.arduino15/packages/esp32/hardware/esp32/2.0.16/cores/esp32/Arduino.h:36,
                 from sketch/ulisp-esp.ino.cpp:1:
/home/fighne/.arduino15/packages/esp32/hardware/esp32/2.0.16/variants/adafruit_qtpy_esp32s2/pins_arduino.h:17: note: this is the location of the previous definition
 #define LED_BUILTIN LED_BUILTIN  // allow testing #ifdef LED_BUILTIN
 
/home/fighne/Documents/GitHub/ulisp/ulisp-esp-4v6/ulisp-esp/ulisp-esp.ino: In function 'object* eval(object*, object*)':
/home/fighne/Documents/GitHub/ulisp/ulisp-esp-4v6/ulisp-esp/ulisp-esp.ino:5756:24: warning: unused variable 'start' [-Wunused-variable]
   static unsigned long start = 0;
                        ^~~~~
/home/fighne/Documents/GitHub/ulisp/ulisp-esp-4v6/ulisp-esp/ulisp-esp.ino: At global scope:
/home/fighne/Documents/GitHub/ulisp/ulisp-esp-4v6/ulisp-esp/ulisp-esp.ino:5756:24: warning: 'start' defined but not used [-Wunused-variable]
/home/fighne/Documents/GitHub/ulisp/ulisp-esp-4v6/ulisp-esp/ulisp-esp.ino: In function 'object* sp_unwindprotect(object*, object*)':
/home/fighne/Documents/GitHub/ulisp/ulisp-esp-4v6/ulisp-esp/ulisp-esp.ino:4245:11: warning: variable 'result' might be clobbered by 'longjmp' or 'vfork' [-Wclobbered]
   object *result;
           ^~~~~~
/home/fighne/Documents/GitHub/ulisp/ulisp-esp-4v6/ulisp-esp/ulisp-esp.ino: In function 'object* sp_ignoreerrors(object*, object*)':
/home/fighne/Documents/GitHub/ulisp/ulisp-esp-4v6/ulisp-esp/ulisp-esp.ino:4277:8: warning: variable 'signaled' might be clobbered by 'longjmp' or 'vfork' [-Wclobbered]
   bool signaled = false;
        ^~~~~~~~
/home/fighne/Documents/GitHub/ulisp/ulisp-esp-4v6/ulisp-esp/ulisp-esp.ino:4268:34: warning: argument 'args' might be clobbered by 'longjmp' or 'vfork' [-Wclobbered]
 object *sp_ignoreerrors (object *args, object *env) {
                          ~~~~~~~~^~~~
Archiving built core (caching) in: /tmp/arduino_cache_114488/core/core_esp32_esp32_adafruit_qtpy_esp32s2_CDCOnBoot_cdc,MSCOnBoot_default,DFUOnBoot_default,UploadMode_cdc,PSRAM_enabled,PartitionScheme_default,CPUFreq_240,FlashMode_qio,FlashFreq_80,FlashSize_4M,UploadSpeed_921600,DebugLevel_none,EraseFlash_none_db46359b928127ee6285ab199180f6a8.a
/home/fighne/.arduino15/packages/esp32/tools/xtensa-esp32s2-elf-gcc/esp-2021r2-patch5-8.4.0/bin/../lib/gcc/xtensa-esp32s2-elf/8.4.0/../../../../xtensa-esp32s2-elf/bin/ld: ulisp-esp.ino.elf section `.dram0.bss' will not fit in region `dram0_0_seg'
/home/fighne/.arduino15/packages/esp32/tools/xtensa-esp32s2-elf-gcc/esp-2021r2-patch5-8.4.0/bin/../lib/gcc/xtensa-esp32s2-elf/8.4.0/../../../../xtensa-esp32s2-elf/bin/ld: region `dram0_0_seg' overflowed by 1984 bytes
collect2: error: ld returned 1 exit status
exit status 1
Error compiling for board Adafruit QT Py ESP32-S2.

This report would have more information with
"Show verbose output during compilation"
option enabled in File -> Preferences.
So changed to memory setting to 8000

fighne, 2024-07-16:
Arduino: 1.8.19 (Linux), Board: "Adafruit QT Py ESP32-S2, Enabled, Disabled, Disabled, Internal USB, Enabled, Default 4MB with spiffs (1.2MB APP/1.5MB SPIFFS), 240MHz (WiFi), QIO, 80MHz, 4MB (32Mb), 921600, None, Disabled"

WARNING: Category '' in library ESP Insights is not valid. Setting to 'Uncategorized'
WARNING: Category '' in library ESP RainMaker is not valid. Setting to 'Uncategorized'
WARNING: Category '' in library WiFiProv is not valid. Setting to 'Uncategorized'
/home/fighne/Documents/GitHub/ulisp/ulisp-esp-4v6/ulisp-esp/ulisp-esp.ino:83: warning: "LED_BUILTIN" redefined
   #define LED_BUILTIN 13
 
In file included from /home/fighne/.arduino15/packages/esp32/hardware/esp32/2.0.16/cores/esp32/esp32-hal-gpio.h:29,
                 from /home/fighne/.arduino15/packages/esp32/hardware/esp32/2.0.16/cores/esp32/esp32-hal.h:83,
                 from /home/fighne/.arduino15/packages/esp32/hardware/esp32/2.0.16/cores/esp32/Arduino.h:36,
                 from sketch/ulisp-esp.ino.cpp:1:
/home/fighne/.arduino15/packages/esp32/hardware/esp32/2.0.16/variants/adafruit_qtpy_esp32s2/pins_arduino.h:17: note: this is the location of the previous definition
 #define LED_BUILTIN LED_BUILTIN  // allow testing #ifdef LED_BUILTIN
 
/home/fighne/Documents/GitHub/ulisp/ulisp-esp-4v6/ulisp-esp/ulisp-esp.ino: In function 'object* eval(object*, object*)':
/home/fighne/Documents/GitHub/ulisp/ulisp-esp-4v6/ulisp-esp/ulisp-esp.ino:5756:24: warning: unused variable 'start' [-Wunused-variable]
   static unsigned long start = 0;
                        ^~~~~
/home/fighne/Documents/GitHub/ulisp/ulisp-esp-4v6/ulisp-esp/ulisp-esp.ino: At global scope:
/home/fighne/Documents/GitHub/ulisp/ulisp-esp-4v6/ulisp-esp/ulisp-esp.ino:5756:24: warning: 'start' defined but not used [-Wunused-variable]
/home/fighne/Documents/GitHub/ulisp/ulisp-esp-4v6/ulisp-esp/ulisp-esp.ino: In function 'object* sp_unwindprotect(object*, object*)':
/home/fighne/Documents/GitHub/ulisp/ulisp-esp-4v6/ulisp-esp/ulisp-esp.ino:4245:11: warning: variable 'result' might be clobbered by 'longjmp' or 'vfork' [-Wclobbered]
   object *result;
           ^~~~~~
/home/fighne/Documents/GitHub/ulisp/ulisp-esp-4v6/ulisp-esp/ulisp-esp.ino: In function 'object* sp_ignoreerrors(object*, object*)':
/home/fighne/Documents/GitHub/ulisp/ulisp-esp-4v6/ulisp-esp/ulisp-esp.ino:4277:8: warning: variable 'signaled' might be clobbered by 'longjmp' or 'vfork' [-Wclobbered]
   bool signaled = false;
        ^~~~~~~~
/home/fighne/Documents/GitHub/ulisp/ulisp-esp-4v6/ulisp-esp/ulisp-esp.ino:4268:34: warning: argument 'args' might be clobbered by 'longjmp' or 'vfork' [-Wclobbered]
 object *sp_ignoreerrors (object *args, object *env) {
                          ~~~~~~~~^~~~
Archiving built core (caching) in: /tmp/arduino_cache_975671/core/core_esp32_esp32_adafruit_qtpy_esp32s2_CDCOnBoot_cdc,MSCOnBoot_default,DFUOnBoot_default,UploadMode_cdc,PSRAM_enabled,PartitionScheme_default,CPUFreq_240,FlashMode_qio,FlashFreq_80,FlashSize_4M,UploadSpeed_921600,DebugLevel_none,EraseFlash_none_db46359b928127ee6285ab199180f6a8.a
/home/fighne/.arduino15/packages/esp32/tools/xtensa-esp32s2-elf-gcc/esp-2021r2-patch5-8.4.0/bin/../lib/gcc/xtensa-esp32s2-elf/8.4.0/../../../../xtensa-esp32s2-elf/bin/ld: ulisp-esp.ino.elf section `.dram0.bss' will not fit in region `dram0_0_seg'
/home/fighne/.arduino15/packages/esp32/tools/xtensa-esp32s2-elf-gcc/esp-2021r2-patch5-8.4.0/bin/../lib/gcc/xtensa-esp32s2-elf/8.4.0/../../../../xtensa-esp32s2-elf/bin/ld: region `dram0_0_seg' overflowed by 704 bytes
collect2: error: ld returned 1 exit status
exit status 1
Error compiling for board Adafruit QT Py ESP32-S2.

This report would have more information with
"Show verbose output during compilation"
option enabled in File -> Preferences.
reducing to 7000 and recompiling

fighne, 2024-07-16:
compiled without error on 2.0.16 with memory setting of 7000
`#elif defined(ARDUINO_ADAFRUIT_QTPY_ESP32S2)
  #define WORKSPACESIZE (7000-SDSIZE)            /* Cells (8*bytes) */
  #define LITTLEFS
  #include <LittleFS.h>
  #define analogWrite(x,y) dacWrite((x),(y))
  #define SDCARD_SS_PIN 13
  #define LED_BUILTIN 13`
will change the core version by version and report when it breaks

fighne, 2024-07-16:
2.0.17 - both 4.4d and 4.6 (with reduced memory setting) compiled successfully
3.0.0 - 4.4d fails with 8000 passes with 7000 and 4.6 with 7000 passes
3.0.1 - both pass with reduced memory setting of 7000
3.0.2 - 4.6 passes output message below
Sketch uses 1053362 bytes (80%) of program storage space. Maximum is 1310720 bytes.
Global variables use 106772 bytes (32%) of dynamic memory, leaving 220908 bytes for local variables. Maximum is 327680 bytes.
3.0.2 - 4.4d passes output message below
Sketch uses 1066506 bytes (81%) of program storage space. Maximum is 1310720 bytes.
Global variables use 105612 bytes (32%) of dynamic memory, leaving 222068 bytes for local variables. Maximum is 327680 bytes.

technoblogy, 2024-07-16:
Is the following summary correct?

* Your original problem was due to including the file `ulisp-extensions.ino`.

Perhaps it would be better to put that in a folder on the repository, with a README.

* The other messages are just warnings.

I'll see if they can be avoided. The `#define LED_BUILTIN 13` may no longer be necessary.

* The error `ld returned 1 exit status` is because the `WORKSPACESIZE` was too large.

I'll check and adjust the memory setting in the next release.

Thank you for the research!

dragoncoder047, 2024-07-16:
> * Your original problem was due to including the file ulisp-extensions.ino.
>
> Perhaps it would be better to put that in a folder on the repository, with a README.

I think a README note would be in order, but in my experience, you can't put other source files in folders or Arduino won't notice them and you'll get "file not found" errors from the compiler.

I haven't tried with .ino files, just .c files, so it might work with those.

technoblogy, 2024-07-16:
Perhaps then I should remove `ulisp-extensions.ino` from the repository completely. It's confusing having it there, and users can download it from the uLisp site anyway.

fighne, 2024-07-16:
So my add would be this the Arduino editor tool, includes all .ino within the folder. So could the file with comments in be renamed .src? Also if the extensions file is needed for all ESP32 why not just include it always?
Final size for version 4.4d with 3.0.2 is 7400
Final size for version 4.6 with 3.0.2 is 7256

technoblogy, 2024-07-17:
> Also if the extensions file is needed for all ESP32 why not just include it always?

No, the extensions file is an optional example, showing how to add extensions. It shouldn't normally be included, but if you do you have to uncomment:

    #define extensions

technoblogy, 2024-07-26:
This should now be fixed.