technoblogy/ulisp-esp #83: `with-spi` missing?

berkowski, 2024-07-18:
I'm looking to play with uLisp on an Adafruit ESP32 feather and a SPI sensor but `with-spi` appears to be undefined.  

There seems to be the hooks for SPI in `ulisp-esp32.ino` but the `with-spi` built-in is not defined.  I went ahead and added `sp_withspi` from the `ulisp-avr.ino` sketch, added a new `string243`, `doc243`, and necessary entry in `lookup_table[]` to `ulisp-esp32.ino`.  After reflashing, `with-spi` seems to work.  Which... is pretty awesome really.

So, is this a mistake in the generation step for building the `ulisp-esp32.ino` source?  Was `with-spi` left out on purpose?  Or am I missing something blatantly obvious?  Apologies if it's the later.  I'm new to uLisp/Arduinos but I've got lots of experience on stm32's with c/c++/rust, so while I've poked around a bit before asking it may be I'm missing something with how the Arduino IDE hides a bunch of the low-level details for sketches.

[patch00-with-spi.txt](https://github.com/user-attachments/files/16287246/patch00-with-spi.txt)

technoblogy, 2024-07-18:
Thank you for pointing this out - it's unintentional, and I'll release an update soon to fix this.

technoblogy, 2024-07-19:
Thank you for pointing this out - it's unintentional, and release 4.6a fixes the omission.

closed 2024-07-19 by technoblogy