technoblogy/ulisp-esp #86: Interopability with idf components/c libraries

Azeirah, 2024-09-19:
Hi! I was looking around for a lisp like language for on an esp. I'm prototyping hardware, and I much prefer the malleable and dynamic nature of a lisp over c.

Now, I do need to use bluetooth. And I read a few pages on the site, but didn't encounter anything.

I'm also assuming that as I'm exploring further, I'd run into more unsupported use-cases.

Does ulisp provide interopability with the "regular" embedded community? Like can I use esp-idf components?

If so, I'd love to explore further! Otherwise I'll stick with plain c for now.

dragoncoder047, 2024-09-19:
uLisp doesn't have any built in Bluetooth module, sorry. And there is no CFFI either. Anyway if you have certain features in espidf that you want to use in uLisp, and you know the C (or preferably Arduino) API, then you can always make an [extension](http://www.ulisp.com/show?19Q4) for it. uLisp is built in to of Arduino which is built on top of espidf.

Also, if you are planning on doing a lot of work on the lisp side vs. C, then get an esp board with PSRAM. This will give you tons more memory for uLisp (with a tradeoff: garbage collections are noticeably slow).

Azeirah, 2024-09-19:
Thank you! I'll stay within the C ecosystem for now, but I'll keep ulisp in the back of my mind.

closed 2024-09-19 by Azeirah